jonlamb-gh/oxcc #12: Determine a reasonable ADC sample time

jonlamb-gh, 2018-08-21:
OSCC was likely using the default Arduino ADC configuration, which could be along the lines of `~104` microseconds per `analogRead()`.

OxCC is currently using a prescaler/cycle-time [combination][] that gives `27.33` microseconds per sample.

[combination]: https://github.com/jonlamb-gh/oxcc/blob/devel/src/board.rs#L45

andrewraharjo, 2018-09-25:
Have you tried to access the F767ZI directly? You can get the read time to 500-700 ns instead use `read()` at 3.5uS. Not sure with the goal you're trying to achieve for sample rate.

jonlamb-gh, 2018-09-25:
Thanks for the suggestions @andrewraharjo. OxCC is currently performing analog reads via basic polling of the ADC channels through the [HAL crate](https://github.com/jonlamb-gh/stm32f767-hal/blob/master/src/adc.rs#L161), used [here](https://github.com/jonlamb-gh/oxcc/blob/devel/src/board.rs#L428).

I was thinking that we should be able to replace the existing OSCC [DAC averaging logic](https://github.com/jonlamb-gh/oscc/blob/master/firmware/common/libs/dac/oscc_dac.cpp#L36), ported [here](https://github.com/jonlamb-gh/oxcc/blob/ca91162b40581827eaa232d1d9c7a7f732a618b1/src/dual_signal.rs#L30), by just using a larger `AdcSampleTime`.

PR #35 (2018-10-04): Migrate to an OSCC compatable ADC configuration
Prior to this commit, the ADCs were configured to run with
full 12 bit conversions. However this can be an inconvenience
for users who wish to use their existing vehicle configuration
data without re-calibrating.

This commit migrates the 12 bit ADC values back to the original
10 bit domain (vehicle constants) and uses the new HAL ADC
conversion resolution parameter for configurations.

Closes #12
Closes #32
closed 2018-10-04 by jonlamb-gh