jonlamb-gh/oxcc #26: Compatibility with Sensor Interface, VCM, and Gateway Boards

andrewraharjo, 2018-09-25:
If my assumption correct, the signal paths for those boards don't change per PolySync released. Is this valid assumption? The Nucleo board outer pinouts serve as Arduino pinouts. I haven't taken a greater look in the code but it does utilize MCP's and DAC's stuffs from Arduino boards.

jonlamb-gh, 2018-09-25:
I think your assumption is pretty reasonable. I didn't see many revisions when looking around in the OSCC repository.

It would be great if OxCC were pin-compatible with existing hardware, but currently it is not.
It does however use most of the same hardware ([MCP 4922 DAC](https://github.com/jonlamb-gh/oxcc/blob/devel/src/dac_mcp4922.rs), etc).

The initial porting effort focused mainly on insuring each module (throttle, brake, and steering) could own whatever peripherals/pins they need access to, such that no module shares a resource with another module.

For example, the analog input pins for each module in [types.rs](https://github.com/jonlamb-gh/oxcc/blob/devel/src/types.rs#L34): ADC1 is given to the brake module, ADC2 to throttle, and ADC3 to steering.

andrewraharjo, 2018-09-26:
Jon,

Do you have the rough schematic to connect one the board with STM32 board? I can help to create new layout for those OSCC boards.

Let me know.

jonlamb-gh, 2018-09-26:
Hi Andrew,

That sounds awesome, thank you!

I put together a rough draw.io schematic of the pins [here](https://github.com/jonlamb-gh/misc-diagrams/blob/master/nucleo-144-oxcc-pins.xml).
[image]

For the CAN transceivers, I have been using the [CAN transceiver SN65HVD230](https://www.waveshare.com/sn65hvd230-can-board.htm) from Waveshare.

andrewraharjo, 2018-09-26:
my pleasure, great work btw. The SN65HVD230 is your CAN controller, isn't it? This might work similar to Kvaser Leaflight I assume. CMIIW

jonlamb-gh, 2018-09-26:
The [CAN bits](https://github.com/jonlamb-gh/stm32f767-hal/blob/master/src/can.rs#L40) I put together for oxcc use the stm's on-board bxCAN controller; which I believe is based on the CAN core of BOSCH's C_CAN controller. AFAICT there are no on-board/integrated transceivers, so I have my CAN1 and CAN2 Rx/Tx pins each wired up to a SN65HVD230 board. For host-side CAN communications, I'm using a PCAN-USB adapter but I have also used the Leaf Lights in the past.

TommyBlairesLewis, 2018-10-05:
@jonlamb-gh Jon, I want to follow up with you regarding this.

I am almost way through to populate the OSCC boards into one dev board. I followed the 1.1.0 design and 0.1.0 for VCM. All hybrid should work. Check this out.

[image]

So, you might have questioned how this is going to compatible with OXCC. I think the my answer is all about dev board

andrewraharjo, 2018-10-05:
@jonlamb-gh 

Check this out! I have made this progress so far. Molex 48 pins connector and getting ready for the VCM and gateway.

[image]

The designs are straight from OSCC, latest build, but at the same time I plan to add a CAN interface to it as well (say CANTact) so the users won't tie using kVaser or any other CAN devices. Although, I will add extra debug port out from the design if the users want to communicate/use kVaser or their own CAN interface.

What do you think? I put this as dev board and you can easily connect them one-one jumper with STM32 development board. Once we figure out the correct wiring, I'll update the board.

jonlamb-gh, 2018-10-05:
Nice work @andrewraharjo, looks like you've made good progress!
It should serve well as a flexible dev board and I'm looking forward to trying one out when ready.

andrewraharjo, 2018-10-05:
Jon, also I’d comment on this. Can we get it work on ROS platform? We should plan to extend this as full ARM-GPU based. 

SoC platform would be better suitable with OXCC rather than tie up with NVIDIA GPU (Jetson, Xavier,etc)

Check this https://www.96boards.org/product/hikey970/

What do you think? Can we support it?

andrewraharjo, 2018-10-05:
@jonlamb-gh 

Nice, that is the direction where it should be. How do you get support on the SoloX Nitrogen6 board? Also, for the higher layer API, what's the plan to make it work with ROS? ROSCCO is all written in C/C++ 

TBH, I am not Rust expert but I need some sort of control command that I can publish/subscribe. If we can figure out, I think that would be my 2nd priority after finishing up the board. Also, if there is any SoC out there that can output the DAC and SPI lines, should we ditch the STM32? I mean I don't mind to create a closed source board with SoC integration. 

Just a matter of time, I guess

jonlamb-gh, 2018-10-05:
Both of those boards have a far amount of community driven support I've found.

You could run [Ubuntu](https://boundarydevices.com/ubuntu-bionic-18-04-lts-for-nitrogen8m-board-september-2018-kernel-4-9-x/), [Yocto](https://boundarydevices.com/nitrogen8m-yocto-gpu-demo/), or some other Linux distro to host your ROS applications.

With these SoC's, ROSCCO is not needed.
Your ROS applications can talk directly to the OxCC firmware via shared memory.
I'm thinking the API could be semantically similar to the existing OSCC API, but backed by a Rust implementation of [RPMsg](https://github.com/OpenAMP/open-amp/wiki/RPMsg-Messaging-Protocol).

I've also had a hard time locating some hardware with the 6 DAC channels OxCC needs, I'll let you know if I come across any.

jonlamb-gh, 2018-10-23:
Nice @andrewraharjo , nothing really stands out to me in your design that would be troublesome, looks good :+1: