jonlamb-gh/oxcc #33: Runtime configuration

jonlamb-gh, 2018-10-03:
It would be handy if users could modify parameters at runtime.

The [bootloader](https://github.com/jonlamb-gh/oxcc/pull/27) I put together already has some configs in flash. It would be easy to reserve a (partial) sector of flash for user firmware configuration parameters.

An initial effort could likely just provide a simple flash get/set framework over CAN; a proper [XCP](https://en.wikipedia.org/wiki/XCP_(protocol)) implementation could be done too.

jonlamb-gh, 2018-10-07:
Posting some links to random open source projects/tools related to this:

- [CCP V2.1](https://automotivetechis.files.wordpress.com/2012/06/ccp211.pdf)
- https://github.com/christoph2/pyxcp
- https://github.com/pySART/pyccp

jed-frey, 2019-03-10:
New link: [XCP Ref Book](https://assets.vector.com/cms/content/application-areas/ecu-calibration/xcp/XCP_ReferenceBook_V3.0_EN.pdf)